Ekumen-OS/andino #93: Wheel size does not match description/controller

BarceloChristian, 2023-07-06:
## Environment
* OS Version: 
* Source or binary build?
  
  

## Description
* Expected behavior:
  - Wheel size is defined the same across all the stack
* Actual behavior:
  - Wheel Diameter is defined to 0.035 for Gazebo's diff drive plugin ([here](https://github.com/Ekumen-OS/andino/blob/humble/andino_gazebo/urdf/andino.gazebo.xacro#L75))
  - Wheel Radius is defined to 0.0331 ([here](https://github.com/Ekumen-OS/andino/blob/humble/andino_description/config/andino/wheel.yaml#L2))

elector102, 2023-07-06:
@BarceloChristian @francocipollone what is the real value?

francocipollone, 2023-07-06:
> @BarceloChristian @francocipollone what is the real value?

The real value of the **Diameter** is 6.6 cm

agalbachicar, 2024-03-19:
@JesusSilvaUtrera will take care right away of this task as he gets to know the platform and this seems to be great issue to begin with.

JesusSilvaUtrera, 2024-03-20:
I have created this [https://github.com/Ekumen-OS/andino/pull/230](PR) for this issue

closed 2024-03-20 by francocipollone