Ekumen-OS/andino #117: The dependencies are not well fulfilled.

elector102, 2023-07-13:
## Environment
* OS Version: ubuntu 22, ros humble
* Source, humble branch

## Description
We are having some dependencies issues using humble branch on a CI. You can see the current dependencies on that image:
[image]

The problem is that andino Gazebo and andino description depend on andino control, and this dependency is fulfilled indirectly.
on the CI we run to compile and then run the test the following:
`colcon build --packages-up-to andino_gazebo aws_robomaker_small_warehouse_world andino_tests --event-handlers console_cohesion+
`
and then:
`colcon test --packages-select andino_gazebo aws_robomaker_small_warehouse_world andino_tests --event-handlers console_cohesion+`
where andino_test are gazebo tests.
With that the dependency of andino gazebo is not fulfilled and CI fail

francocipollone, 2023-07-13:
Thanks for filling up the issue @elector102 : Check https://github.com/Ekumen-OS/andino/pull/118

francocipollone, 2023-07-13:
> CI fail

What is the issue?

Now I am realizing that it is ok that `andino_control` is not a dependency to `andino_gazebo`. I think that the dependency that is missing is `controller_manager`

francocipollone, 2023-07-13:
> > CI fail
> 
> What is the issue?
> 
> Now I am realizing that it is ok that `andino_control` is not a dependency to `andino_gazebo`. I think that the dependency that is missing is `controller_manager`

Oh yeat it is being used when finding the controller's parameters

https://github.com/Ekumen-OS/andino/blob/5236b1132343539e19c2eb63c61e5f5f7814555b/andino_gazebo/urdf/andino.gazebo.xacro#L48

agalbachicar, 2023-07-13:
There is another occurency:

https://github.com/Ekumen-OS/andino/blob/humble/andino_description/urdf/andino.urdf.xacro#L84-L87

That imports files from `andino_control` and if `andino_description` includes `andino_control` as a dependency, then you have a circle one.

#118 deals with the issue, but the urdf files need to be ordered to solve this problem.

francocipollone, 2023-07-13:
> There is another occurency:
>
> https://github.com/Ekumen-OS/andino/blob/humble/andino_description/urdf/andino.urdf.xacro#L84-L87
>
> That imports files from `andino_control` and if `andino_description` [...]

```xml
<xacro:include filename="$(find ${package_name})/urdf/include/andino_control.urdf.xacro" />
```

Here `package_name` unpacks to `andino_description` so it isn't looking for stuff in `andino_control` package.
So I don't see the cycling dependency.

**It is true though** that if we are strict we could move the `andino_control.xacro.urdf` to `andino_control` package in order to extend the base description to ros control if needed.

However, based on taking a look at other robots descriptions, in general they add the ros control stuff within the base urdf description and it is quite accepted.

francocipollone, 2023-08-02:
Pinging @agalbachicar to define the bug if any. I will close this otherwise

PR #118 (2023-07-13): Adds andino_control as dependency of andino_gazebo
# 🦟 Bug fix

Fixes #117 

## Summary
What the title says

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if it affects the public API)
closed 2023-10-11 by francocipollone