Ekumen-OS/andino #134: L298N wiring diagram not correct

Russ76, 2023-08-05:
Do you really have your robot driving around with wiring like in the diagram under hardware?

For it appears somewhat like the wiring for a stepper motor, but L298N can only control one stepper motor!

For powering two motors, with variable speeds, the wiring connections are different.

https://tronixstuff.com/2014/11/25/tutorial-l298n-dual-motor-controller-modules-and-arduino/

Also, the 5V terminal is for output, not for input. The Voltage regulator on the module can supply 5V to Arduino or encoders if properly wired.

Russ

## Environment
* OS Version: 
* Source or binary build?
  
  

## Description
* Expected behavior: 
* Actual behavior: 

## Steps to reproduce

1.
2.
3.

## Output

francocipollone, 2023-08-06:
Thanks for commenting, let me go through your comment.

> For it appears somewhat like the wiring for a stepper motor, but L298N can only control one stepper motor!

L298N can control one stepper motor or two dc motors. The latter is our case, so we are ok.

> For powering two motors, with variable speeds, the wiring connections are different.

Base on what you say that the connection is different? Can you elaborate?

> Also, the 5V terminal is for output, not for input.

That's depend, you can remove the jumper (as indicated in the picture) and avoid using the internal 5v regulator to supply directly the current for the l298.

> The Voltage regulator on the module can supply 5V to Arduino or encoders if properly wired.

That's a decision yes, you can go that path. The diagram is a suggestion that works but the end design is open to the user.
For example, adding a step up to raise to 9V from 5V the voltage of the motor (at the l298 input) is a recommendation to have better reaction from the motors.

Russ76, 2023-08-06:
For powering two motors, the enable pins are necessary and are used for PWM for motors. The other two pair are for direction control and are only digital high or low. It looks to me like you have wiring connections correct for the BTS controller, which uses 2 PWM wires for each motor, one for forward, and one for reverse.

closed 2023-12-07 by francocipollone