Ekumen-OS/andino #135: Allow to load custom configuration for the description

francocipollone, 2023-08-16:
### Context

Nowadays the configuration files that describe the robot are located at:
https://github.com/Ekumen-OS/andino/tree/humble/andino_description/config/andino

However if the user wanted to have it a bit different; `motors` added in a different manner, `lidar` displaced and connected somwhere else, `base` with different size; they should create their own URDF files as the configuration files are harcoded within the [`xacro` file](https://github.com/Ekumen-OS/andino/blob/humble/andino_description/urdf/andino.urdf.xacro)

### Proposal
- Modify the xacro to receive by argument the path to the configuration files
- Use as default the current files if no argument is passed down to the xacro files.
- Modify the launch file to accept passing the path to the configuration yaml.

JesusSilvaUtrera, 2024-03-26:
@francocipollone We should close this issue if it is already solved and merged

jballoffet, 2024-05-08:
@lneumarkt, are you currently working on this? No worries if not, just let us know since @JesusSilvaUtrera was considering picking it up. Thanks!

lneumarkt, 2024-05-08:
I'm not working on is. Feel free to pick it up

jballoffet, 2024-05-08:
Got it, thanks @lneumarkt for the prompt response!

JesusSilvaUtrera, 2024-05-09:
Thanks @lneumarkt for the response, I will take this issue.

PR #243 (2024-05-15): Allow custom configuration
# 🎉 New feature

Closes #135  and #136 

## Summary
I have continued the work on these issues to add the possibility to pass arguments to the xacro file directly from the launch file. I created a `hardware.yaml` in the `config` folder to load the params from there instead of hardcoding them in the `andino_control.xacro` file.

I have also updated the `README.md` with these new features.

## Test it
You can test that it works running `ros2 launch andino_description view_andino.launch.py`, which executes RViz and the `robot_state_publisher` to check that everything rem [... cut]
closed 2024-05-15 by jballoffet