Ekumen-OS/andino #136: Move hardware interface plugin configuration to a config file.

francocipollone, 2023-08-16:
Depends on #135 

### Context
The configuration of the andino hardware interface is harcoded in the urdf

https://github.com/Ekumen-OS/andino/blob/70a9ad6d65089c1b999859ed6a5113fb0d23bdd0/andino_description/urdf/include/andino_control.urdf.xacro#L6-L15

### Proposal
 - Request a configuration yaml for the ros control configuration

JesusSilvaUtrera, 2024-05-09:
As this issue is related to #135, I will take it as well ans try to submit a PR with both issues.

JesusSilvaUtrera, 2024-05-09:
@francocipollone Do you think it would be better to move the `andino_control.urdf.xacro` to `andino_control`? I think it is better to have all the xacro files together, if we do that we would create a dependency between `andino_description` and `andino_control`, and also we would need to add dependencies like xacro to `andino_control`.

francocipollone, 2024-06-09:
> @francocipollone Do you think it would be better to move the `andino_control.urdf.xacro` to `andino_control`? I think it is better to have all the xacro files together, if we do that we would create [...]

Let's keep it in andino_description for now, thanks for checking

JesusSilvaUtrera, 2024-06-14:
@francocipollone I just noticed that this was merged but the issue was not closed, JIC you want to close it.

closed 2024-06-19 by francocipollone