Ekumen-OS/andino #148: Velocity not limited for the diff drive controller

IDavGal, 2023-09-04:
## Description
The [`andino_controllers.yaml`](https://github.com/Ekumen-OS/andino/blob/humble/andino_control/config/andino_controllers.yaml) configuration file specify no limits in the linear/angular velocity. According the [motor's description](https://www.sparkfun.com/products/16413) its max velocity is 90 RPM, given the wheel radius this would translate to roughly 0.33 m/s linear vel. 
The `diff_drive_controller` handles both [velocities limits separately](https://github.com/ros-controls/ros2_controllers/blob/1dac309979fb9392dc12feb6b46555582fed6fc6/diff_drive_controller/src/diff_drive_controller.cpp#L237C2-L237C2), so the border case in which `linear_vel + angular_vel` translates to a higher motor velocity than 90 RPM is not considered by this calculation.

IDavGal, 2023-10-04:
There is an open PR in [`ros2_controllers`](https://github.com/ros-controls/ros2_controllers/pull/471) addressing this issue, although is currently pause.
I'll try to either bump it up or address the issues myself.

francocipollone, 2023-10-04:
> There is an open PR in [`ros2_controllers`](https://github.com/ros-controls/ros2_controllers/pull/471) addressing this issue, although is currently pause. I'll try to either bump it up or address the issues myself.

I should run a local test in the real robot and provide a real-based approximation, wdyt?
My estimation is that we are going to be below 0.5 m/s without effort.