Ekumen-OS/andino #152: Rename andino_gazebo package to andino_gz_classic

francocipollone, 2023-09-21:
### Context
- gazebo classic -> https://classic.gazebosim.org/
- gazebo -> https://gazebosim.org/home

### Summary

In order to keep a correct nomenclature we should rename the `andino_gazebo` package and fix all the appearances of this.
The simulation we have here isn't using the `gazebo` simulator but the `gazebo classic` one. In order to allow a `gazebo` simulator to happen we need to fix this colliding name.

### Options

- andino_gz_classic
- andino_gazebo_classic  -> Probably too long

In the future a `gazebo` simulation will be:
- andino_gz

francocipollone, 2023-09-21:
@jballoffet any preference?

jballoffet, 2023-09-21:
andino_gz and andino_gz_classic SGTM!

agalbachicar, 2023-09-21:
This is a breaking change for https://github.com/Ekumen-OS/andino_integration_tests
Not blocking, just tracking :)

francocipollone, 2023-09-21:
> This is a breaking change for https://github.com/Ekumen-OS/andino_integration_tests Not blocking, just tracking :)

Yeah I got you guys in mind

PR #153 (2023-09-27): Renames andino_gazebo package to andino_gz_classic.
# Renaming package

Closes #152 

## Summary
 - Renames `andino_gazebo` package to `andino_gz_classic`

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if it affects the public API)
closed 2023-09-27 by francocipollone