Ekumen-OS/andino #169: Adds camera_info yaml file

francocipollone, 2023-10-30:
### Summary

When launching the camera we are not passing a yaml with the description of the camera's intrinsics

https://github.com/Ekumen-OS/andino/blob/024b6943e5813c2b555b3f74e0e4e97f8ed21cba/andino_bringup/launch/camera.launch.py#L34-L46

### Notes

- https://navigation.ros.org/tutorials/docs/camera_calibration.html#overview
- https://index.ros.org/r/v4l2_camera/
- https://github.com/tier4/ros2_v4l2_camera/blob/87d0ff0d24302124ee9a4b8e00ad6d675446b2e3/launch/v4l2_camera.launch.py#L119-L120

### Definition of done

 - [ ] Adds a default yaml file descringin camera intrinsic.
 - [ ] Adds documentation about how to calibrate the camera if needed.

JesusSilvaUtrera, 2024-03-20:
I will take care of this issue

JesusSilvaUtrera, 2024-03-20:
I have created this [PR](https://github.com/Ekumen-OS/andino/pull/231) @agalbachicar

PR #231 (2024-06-17): Jesus/#169 camera info yaml
# 🦟 Bug fix

Closes #169 

## Summary
I have added a YAML file with the default camera intrinsics to load into the camera.launch.py file. I also included as a comment the link to the info of the camera, just in case we need more technical parameters of it. 

The path to the YAML file is included as a parameter in the launch file, just in case it needs to be modified.
closed 2024-06-17 by jballoffet