Ekumen-OS/andino #170: Enable compressed image transport

francocipollone, 2023-10-30:
### Summary

Nowadays the images from the camera are only raw images

### Notes

 - Simply installing the correspondent image transport plugijn will enable the compressed image to be also published
 - In order to visualize images in RViz2 you will need to uncompress the image as it natively doesn't support compressed images to be viewed.
 
### Definition of done

- [ ] Provide compressed image transport
- [ ] Adds documentation about compressed image visualization

JesusSilvaUtrera, 2024-03-20:
I will take care of this issue

JesusSilvaUtrera, 2024-03-20:
I have reviewed this, and the package 'ros-humble-image-transport-plugins' is installed automatically when the 'rosdep' command is used, as well as its dependencies (```sudo apt install libtheora-dev libogg-dev libboost-python-dev```), so there are no changes needed there.

On the other side, as 'ros-humble-image-transport' is also installed when running 'rosdep', we can use the 'republish' node to uncompress the image and see it in RViz:

```sh
ros2 run image_transport republish raw --ros-args --remap in:=/image_raw/compressed --remap out:=/image_raw/uncompressed
```

The syntax used for this node is different than the one @francocipollone used, because it says that syntax is deprecated.

For evidence, you can see the output of the node 'list_transports':

```sh
jesus-silva@themis:~/ws$ ros2 run image_transport list_transports 
Declared transports:
image_transport/compressed
image_transport/compressedDepth
image_transport/raw
image_transport/theora

Details:
----------
"image_transport/compressed"
 - Provided by package: compressed_image_transport
 - Publisher: 
      This plugin publishes a CompressedImage using either JPEG or PNG compression.
    
 - Subscriber: 
      This plugin decompresses a CompressedImage topic.
    
----------
"image_transport/compressedDepth"
 - Provided by package: compressed_depth_image_transport
 - Publisher: 
      This plugin publishes a compressed depth images using PNG compression.
    
 - Subscriber: 
      This plugin decodes a compressed depth images.
    
----------
"image_transport/raw"
 - Provided by package: image_transport
 - Publisher: 
      This is the default publisher. It publishes the Image as-is on the base topic.
    
 - Subscriber: 
      This is the default pass-through subscriber for topics of type sensor_msgs/Image.
    
----------
"image_transport/theora"
 - Provided by package: theora_image_transport
 - Publisher: 
      This plugin publishes a video packet stream encoded using Theora.
    
 - Subscriber: 
      This plugin decodes a video packet stream encoded using Theora.
```

I will try to test this with my webcam as I don't have the Andino at home to check all works without problems.

JesusSilvaUtrera, 2024-03-20:
I have been struggling for a while to open my webcam using image_transport:

```sh
[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ./modules/videoio/src/cap_v4l.cpp (890) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ros2run]: Process exited with failure 1
```

When I perform a ```ls /dev/video*´´´ the webcam is found, but I think docker is not able to access it inside the container, since the container is not being run with 'privileged' access. I have tried with an app like 'cheese' and indeed, the webcam is not being accessed even though it is detected, so I can't continue with this.

JesusSilvaUtrera, 2024-03-20:
@agalbachicar to keep you updated.

francocipollone, 2024-03-20:
> When I perform a ```ls /dev/video*´´´ the webcam is found, but I think docker is not able to access it inside the container, since the container is not being run with 'privileged' access. I have [...]

Are you using the docker provided in this repo? It is actually running with `--privileged` flag

https://github.com/Ekumen-OS/andino/blob/humble/docker/run.sh#L107-L116

Probably you could try also adding a bind volume `-v` to /dev. You can modify the run.sh script locally for trying that

JesusSilvaUtrera, 2024-03-20:
> Are you using the docker provided in this repo? It is actually running with `--privileged` flag
>
> https://github.com/Ekumen-OS/andino/blob/humble/docker/run.sh#L107-L116
>
> Probably you could try also [...]

Yeah I am using the docker provided, so if it uses the ```--privileged``` flag by default it is a strange behavior. I will try what you suggests later, thanks Franco!

JesusSilvaUtrera, 2024-03-21:
@francocipollone trying with the new volume worked, now I can see my webcam inside the docker container and I can publish the images as ROS msg.

The results are that indeed if the plugins are installed (and they are installed by default after `rosdep`) these topics are published:

[image]

So as you can see both the compressed and the raw image are published, so there is no need to 'republish' the compressed image as raw to see it in RViz, you can simply use the raw image in RViz. 

The problem with this is the bandwidth, because it would be more convenient if we could simply publish only the formats we want (in this case the raw and compressed, the others are not important). 

I will look into that and be back here

JesusSilvaUtrera, 2024-03-21:
I have tried it, and although the topic is being republished with `sensor_msgs/msg/Image` type, it is not possible to see it on RViz:

[image]

On the other hand, I have looked if it is possible to publish only some of transports, and the only information I have found is [this](https://github.com/ros-perception/image_transport_tutorials?tab=readme-ov-file#adding-new-transports) so far.

JesusSilvaUtrera, 2024-03-22:
@agalbachicar in case you want to check the updates :)

JesusSilvaUtrera, 2024-05-24:
@francocipollone I think we can close this issue given the results:

- The compressed image is being published automatically when you have the plugins installed, and the node also publishes the original image uncompressed.
- Rviz2 still doesn't support compressed images visualization, but we can use the original uncompressed `/camera/image` topic.

Let me know what you think and if you want to test anything else now that I have the Raspy Cam.

francocipollone, 2024-06-09:
Sorry for the late reply. 
Thanks for the investigation @JesusSilvaUtrera, it is much appreciated! 
We can close this

closed 2024-06-09 by francocipollone