Ekumen-OS/andino #174: Terminate called after throwing an instance of 'std::runtime_error'

mlde50, 2023-11-19:
## Environment
* OS Version: 
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy
```
* ROS2 Humble updated
* Andino - Commits on Oct 9, 2023

## Description
* Check motor with motor_driver_demo
* 
* 
* 
*  Expected behavior: 
```
 motor_driver_demo --msg='o 255 255'
``` 
* Actual behavior: 
```$ motor_driver_demo --msg='o 255 255'
Motor driver is connected: True
Waiting 2 seconds for the Microcontroller to be ready...
Sending message: o 255 255
Response to o 255 255 timed out.
Response: 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Input/output error
Aborted (core dumped)
```
## Steps to reproduce
1.
```
$ motor_driver_demo --serial_port=/dev/ttyUSB0 --msg='e'
```
2.
```
$ motor_driver_demo --msg='o 255 255'
```

## Output
1.
```
Motor driver is connected: True
Waiting 2 seconds for the Microcontroller to be ready...
Sending message: e
Response: 0 0
```
2.
```
Motor driver is connected: True
Waiting 2 seconds for the Microcontroller to be ready...
Sending message: o 255 255
Response to o 255 255 timed out.
Response: 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Input/output error
Aborted (core dumped)
```

francocipollone, 2023-11-21:
Hey @mlde50 thanks for filling this in, I will try to replicate this locally whenever I have a chance. 
In the meantime, just in case, have you checked the configuration part at https://github.com/Ekumen-OS/andino/tree/humble/andino_hardware#arduino ?

francocipollone, 2023-11-21:
@mlde50  Note that the `motor_driver_demo` app is just sending messages to the Arduino. You can try sending those very same commands via Arduino IDE so we can narrow down the issue and check if your connection with the Arduino is working properly.

After loading the firmware into the Arduino uno/nano platform you can open the Serial Monitor, indicate the Baud Rate(57600) and  CR (carriage return)

[image]

 and send:
- `e` --> for getting the encoders values
- `o 255 255`

This is a bit explained at https://github.com/Ekumen-OS/andino/tree/humble/andino_firmware#andino_firmware

mlde50, 2023-11-22:
Hey @francocipollone, thanks for your reply. I did more investigation with latest commit and Arduino monitor application tool. And I have some strange behavior. When power on arduino Nano, "e" command return encoder value 
[image]

But when I send "o 255 255" command motor doesn't rotate and communication between computer and Arduino is lost, impossible to resend now command without unplug and plugin arduino.

mlde50, 2023-11-25:
Hey @francocipollone, I reinstall every things and it's seems ok  now.

closed 2023-11-25 by mlde50