Ekumen-OS/andino #182: [ERROR] [ros2_control_node-3]: process has died

mlde50, 2023-11-25:
## Environment
* OS Version:
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy
```

* ROS2 Humble updated
* Andino - latest commits

## Description
* ros2 launch andino_bringup andino_robot.launch.py include_rplidar:=false include_camera:=false
* Actual behavior: 
```
... Bad file descriptor
[ros2_control_node-3] terminate called after throwing an instance of 'LibSerial::NotOpen'

...

 Bad file descriptor
[ros2_control_node-3] terminate called after throwing an instance of 'LibSerial::NotOpen'

```

## Steps to reproduce
1.
```
colcon build
[1.133s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/mlde/rosws/install/andino_slam' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/mlde/rosws/install/andino_navigation' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/mlde/rosws/install/andino_hardware' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/mlde/rosws/install/andino_gz_classic' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/mlde/rosws/install/andino_firmware' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/mlde/rosws/install/andino_bringup' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/mlde/rosws/install/andino_control' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/mlde/rosws/install/andino_description' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/mlde/rosws/install/andino_base' in the environment variable AMENT_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mlde/rosws/install/andino_slam' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mlde/rosws/install/andino_navigation' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mlde/rosws/install/andino_hardware' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mlde/rosws/install/andino_gz_classic' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mlde/rosws/install/andino_firmware' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mlde/rosws/install/andino_bringup' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mlde/rosws/install/andino_control' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mlde/rosws/install/andino_description' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[1.133s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mlde/rosws/install/andino_base' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> andino_base
Starting >>> andino_description
Starting >>> andino_firmware                                                      
Starting >>> andino_hardware
Starting >>> andino_slam                                                                                                            
Finished <<< andino_firmware [1.33s]                                                                                                                                                   
Finished <<< andino_hardware [1.44s]                                                                                                                          
Finished <<< andino_slam [1.45s]                                                                                             
Finished <<< andino_description [1.64s]
Finished <<< andino_base [6.04s]                     
Starting >>> andino_control
Finished <<< andino_control [0.73s]                  
Starting >>> andino_gz_classic
Starting >>> andino_bringup
Finished <<< andino_bringup [0.81s]                                                                 
Finished <<< andino_gz_classic [0.91s]
Starting >>> andino_navigation
Finished <<< andino_navigation [0.76s]                  

Summary: 9 packages finished [9.45s]
[... 67 lines cut ...]
[ros2_control_node-3] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f0b78e42475]
[ros2_control_node-3] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-3]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-3]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f0b78e969fc]
[ros2_control_node-3] Aborted (Signal sent by tkill() 89212 1000)
[ERROR] [ros2_control_node-3]: process has died [pid 89212, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_l0excwdc --params-file /home/mlde/rosws/install/andino_control/share/andino_control/config/andino_controllers.yaml'].
[spawner-4] [INFO] [1700910621.319929034] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1700910623.335623599] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1700910625.348322378] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [INFO] [1700910627.362411328] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-4] [ERROR] [1700910629.379570866] [spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner-4]: process has died [pid 89214, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --ros-args'].
[INFO] [spawner-5]: process started with pid [89236]
[spawner-5] [INFO] [1700910631.774598994] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1700910633.790682407] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1700910635.805946622] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [INFO] [1700910637.820584740] [spawner_diff_controller]: Waiting for '/controller_manager' node to exist
[spawner-5] [ERROR] [1700910639.835906287] [spawner_diff_controller]: Controller manager not available
[ERROR] [spawner-5]: process has died [pid 89236, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner diff_controller --controller-manager /controller_manager --ros-args'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1700911390.320227209] [rclcpp]: signal_handler(signum=2)
[twist_mux-2] [INFO] [1700911390.320261052] [rclcpp]: signal_handler(signum=2)
[INFO] [twist_mux-2]: process has finished cleanly [pid 89182]
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 89180]
```

francocipollone, 2023-11-28:
It isn't being able to connect with the Arduino. It can't open the serial port.

Have you followed all the instructions from here? Take a look at https://github.com/Ekumen-OS/andino/tree/humble/andino_hardware#single-board-computer-sbc

Note there are some configurations for the port names.

francocipollone, 2023-11-28:
All the instructiosn are important but for this particular error basically make sure you've followed the sections:
 - https://github.com/Ekumen-OS/andino/tree/humble/andino_hardware#arduino
 - https://github.com/Ekumen-OS/andino/tree/humble/andino_hardware#usb-port-name-configuration

mlde50, 2023-11-29:
Hi @francocipollone , thanks for the reply.  There is a good documentation, and I following the steps rigorously. I going to retry later.

francocipollone, 2023-12-07:
Closing this issue as resolved. In case you have a followup question or inquiry while building your Andino feel free to comment at QA category under the discussion section: https://github.com/Ekumen-OS/andino/discussions/categories/q-a

closed 2023-12-07 by francocipollone