Ekumen-OS/andino #190: Improve andino_navigation package

francocipollone, 2023-12-15:
### Summary

Improve `andino_navigation` package to make it easier to the user to use nav2 on the real robot as well as to decouple `andino_navigation` from the simulation.

### Tasks:
 - [ ] Test `andino_navigation/launch/bringup.launch.py` launch file with real robot. 
       - Tune parameters for a better experience.
       - Note odom topic is different in simulation and in real robot. (Should we here try to standarize it?)
         - Temporarily having two param files might be ok.
 - [ ] Include nav bringup at a `andino_bringup` package level so it is easier to launch everything.
 - [ ] Move `andino_navigation/launch/andino_simulation_navigation.launch.py` file to `andino_gz_classic` package as it depends on gazebo.

agalbachicar, 2024-02-06:
If I may:

>  Include nav bringup at a andino_bringup package level so it is easier to launch everything.

Keep it separate and be more atomic with your package dependencies. If you just want to have basic sensor / actuator functionality, the nav stack would make it bring a plethora of dependencies you don't want.

> Move andino_navigation/launch/andino_simulation_navigation.launch.py file to andino_gz_classic package as it depends on gazebo.

Actually, one of the things I struggled the most with is the coupling in the packages. I would highly recommend to have launchfiles that are specific to the nav stack than only care about loading the nodes that are required for that and then demos / integration packages that would put pieces together. That would help also to make sure you can select, as a user, where to launch things (physical machine / docker container).

francocipollone, 2024-02-13:
> Keep it separate and be more atomic with your package dependencies. If you just want to have basic sensor / actuator functionality, the nav stack would make it bring a plethora of dependencies you don't want.

Thanks for the suggestion. Agree with you about the dependencies nav2 would bring at `andino_bringup` package

closed 2024-02-14 by francocipollone