Ekumen-OS/andino #192: Migrate to modern Gazebo

azeey, 2023-12-21:
## Desired behavior
As you may know, Gazebo-classic (aka Gazebo11, see [Terminology](https://gazebosim.org/about#:~:text=Terminology)) is no longer being actively developed  and will not be available on ROS Jazzy ([REP 2000](https://www.ros.org/reps/rep-2000.html#jazzy-jalisco-may-2024-may-2029)). I noticed that this project still uses Gazebo-classic. Are there any plans to migrate to modern Gazebo ([previously known as Ignition](https://community.gazebosim.org/t/a-new-era-for-gazebo/1356))?

## Alternatives considered
n/a

## Implementation suggestion
Steps for Migration:

- Follow https://gazebosim.org/api/sim/8/migrationsdf.html to update SDF files
- Follow https://gazebosim.org/api/sim/8/migrationplugins.html to update custom plugins 
- Update launch files to use [ros_gz](https://github.com/gazebosim/ros_gz/) instead of gazebo_ros_pkgs. Documentation for ros_gz can be found at https://gazebosim.org/docs/harmonic/ros2_integration. Another good resource is the [ros_gz_project_template](https://gazebosim.org/docs/harmonic/ros_gz_project_template_guide#guide-to-code-class-codespan-ros_gz_project_template-code-for-ros-2-and-gazebo-development).

## Additional context
n/a

francocipollone, 2023-12-21:
Hey @azeey thanks for the issue and suggestion in the implementation :smile: , it is useful indeed.
Happy to mention that it is already part of our backlog and it is under migration. We haven't opened it yet but we will do so shortly. (It isn't completely migrated I think we are still missing the sensors)

This won't replace the current gz_classic simulation but will provide an alternative, similar to what we did with [andino_webots](https://github.com/Ekumen-OS/andino_webots)

azeey, 2023-12-21:
That's great to hear!vLet me know if you run into any issues.

francocipollone, 2023-12-21:
> That's great to hear!vLet me know if you run into any issues.

Thanks!, I'll keep you posted! :rocket:

francocipollone, 2024-02-02:
- :heavy_check_mark:  Tackled at https://github.com/Ekumen-OS/andino_gz
- :heavy_check_mark:  Referenced at the readme: https://github.com/Ekumen-OS/andino?tab=readme-ov-file#paperclips-related-projects

closed 2024-02-02 by francocipollone