Ekumen-OS/andino #198: Camera from RPI has a driver issue in Ubuntu Mate 22.04 which prevents proper loading

agalbachicar, 2023-12-28:
## Environment
* OS Version: RPI 4, Ubuntu Mate 22.04.3, humble
* Source or binary build: source build, [2e552b5](https://github.com/Ekumen-OS/andino/commit/2e552b51e3333d6e2c6186defe8b8a15ad370f9d)

## Description
* Expected behavior: after running the installation guide in the RPI, and launching the base launchfile the camera would be up and working.
* Actual behavior: it does not work, fails with: 
```
[v4l2_camera node-3] [ERROR] [1703759493.768683193] [v4l2_camera]: Failed stream start: Invalid argument (22)
```

## Steps to reproduce

1. Run the installation instructions in https://github.com/Ekumen-OS/andino/blob/humble/andino_hardware/README.md
2. Go to https://github.com/Ekumen-OS/andino/blob/humble/README.md#rocket-usage and try:
```
ros2 launch andino_bringup andino_robot.launch.py
```
The launch execution would fail with that error listed above.

## Notes

I checked and it seems this is a known error:

- https://robotics.stackexchange.com/questions/23967/raspberry-pi-ros-2-humble-on-ubuntu-22-04-server-v4l2-camera-reports-failed
- https://gitlab.com/boldhearts/ros2_v4l2_camera/-/issues/45

It seems there is a workaround with raspian distros, however I did not find the `/boot/firmware/config.txt` or any similar file under the `/boot` directory nor that configuration in any other file (`camera_auto_detect`).

## Hardware

- Rasberry PI camera module 2
- Raspberry PI 4 B 8GB

flsmr, 2024-05-19:
For me, the following fix got my raspi cam v2 work again (Ubuntu Mate 22.04 on Raspberry Pi 4B):
change/add the following lines within /boot/firmware/config.txt:

```
# Autoload overlays for any recognized cameras or displays that are attached
# to the CSI/DSI ports. Please note this is for libcamera support, *not* for
# the legacy camera stack
start_x=1
gpu_mem=128
# camera_auto_detect=1
```

francocipollone, 2024-05-19:
> For me, the following fix got my raspi cam v2 work again (Ubuntu Mate 22.04 on Raspberry Pi 4B): change/add the following lines within /boot/firmware/config.txt:
>
> ```
> # Autoload overlays for any [...]

Thanks for sharing the fix @flsmr

JesusSilvaUtrera, 2024-05-27:
> For me, the following fix got my raspi cam v2 work again (Ubuntu Mate 22.04 on Raspberry Pi 4B): change/add the following lines within /boot/firmware/config.txt:
>
> ```
> # Autoload overlays for any [...]

Exactly, I have tried my raspi cam v2 today and this was the solution to make it work, apart from enabling it in the raspberry configuration:

```
sudo raspi-config
```

Go to `Interface Options`, select `camera` and enable it. Then you just need to reboot and the camera should be working fine.

francocipollone, 2024-05-27:
We should update this fix to the instructions in case others face this very same issue.

JesusSilvaUtrera, 2024-05-27:
> We should update this fix to the instructions in case others face this very same issue.

I'm on it, it will be included in the #231 PR