Ekumen-OS/andino #200: Provide a dockerized environment for ARM64

agalbachicar, 2023-12-28:
## Desired behavior

Reduce the time to deploy an Andino bot in a fully functional system.

## Alternatives considered

Probably there are many out there, like from a bash script, having an image of the SD card, etc. etc. etc.
However, I think that starting by having a dockerized version that can be built in a host machine (x86) that mimics the build procedure in the final architecture (arm64) allows to catch issues with incompatible processes and test many things in CI (not yet, Github Actions doesn't support arm64 yet).

## Implementation suggestion

I would consider building in stages and providing multiple images:

- Image for development in RPI, but with no workspace specific dependency. (A)
- Image that clones the repo, installs the workspace, builds and tests the packages in andino. (B)
- Image that mounts the repo as a volume and it is suitable for continuous development. (C)

B depends on A.
C depends on A.

A and B can be distributed via Github, probably.  

## Additional context

Running https://github.com/Ekumen-OS/andino/tree/humble/andino_hardware is tedious. At least, we can reduce the burden by providing the specifics of software packages installed in a docker container.

agalbachicar, 2023-12-28:
Connected with https://github.com/Ekumen-OS/andino/discussions/20

agalbachicar, 2024-01-02:
Connected to #51

marcoshuck, 2024-01-03:
https://docs.docker.com/build/ci/github-actions/multi-platform/ In case this becomes relevant. You can set up QEMU to build arm64.