Ekumen-OS/andino #201: Fix rosdep dependencies when installing in Raspeberry Pi

agalbachicar, 2023-12-29:
## Environment
* OS Version: Ubuntu 22.04 / Ubuntu Mate 22.04 - arm64/v8
* Source build: [9f7793d](https://github.com/Ekumen-OS/andino/commit/9f7793db6810d6d4deeca745ef18aeb86660a444)

## Description
* Expected behavior: running rosdep as listed in the guide does not lead to any error.
* Actual behavior: because the repository has the simulation packages, it fails to resolve all gazebo related dependencies and also one related to the turtlebot3 (potentially unused).

## Steps to reproduce

1. Run rosdep as described above and in the readmes when running in the RPI.

```
rosdep install --from-paths src --ignore-src -i -y
```

## Output

```
rosdep install --from-paths src --ignore-src -i -y
[...]
ERROR: the following rosdeps failed to install                                            
   apt: command [sudo -H apt-get install -y ros-humble-turtlebot3-gazebo] failed           
   apt: command [sudo -H apt-get install -y ros-humble-gazebo-ros-pkgs] failed             
   apt: command [sudo -H apt-get install -y ros-humble-gazebo-ros] failed                  
   apt: command [sudo -H apt-get install -y ros-humble-gazebo-ros2-control] failed         
   apt: Failed to detect successful installation of [ros-humble-turtlebot3-gazebo]         
   apt: Failed to detect successful installation of [ros-humble-gazebo-ros-pkgs]           
   apt: Failed to detect successful installation of [ros-humble-gazebo-ros]                
   apt: Failed to detect successful installation of [ros-humble-gazebo-ros2-control]
```

agalbachicar, 2023-12-29:
I suspect we need to move the simulation specific packages to another repository to solve this issue.

robofoundry, 2024-01-27:
I ran into same issue and as instructions suggest you can continue installing remaining packages by using rosdep with -r switch OR you can also add COLCON_IGNORE empty file in package root folder you want to ignore. In this case it would be gz_classic, nav and slam packages. Or you can simply delete those three package directories on robot/RPi. HTH.

JesusSilvaUtrera, 2024-09-17:
@jballoffet I suppose we can also close this one, as using the `-r` flag with `rosdep` command solves the problem and also on Jazzy we have deprecated the simulation packages.