Ekumen-OS/andino #228: Remove simulation dependency from andino_navigation

garyservin, 2024-02-14:
## Desired behavior
The andino_navigation shouldn't depend on sim packages https://github.com/Ekumen-OS/andino/blob/humble/andino_navigation/package.xml#L17

## Implementation suggestion
The `andino_simulation_gazebo.launch.py` should probably be moved to the `andino_gz_classic` pacakge

francocipollone, 2024-03-20:
Sorry for the late reply @garyservin ! 
Indeed that dependcy isn't ideal.

andino_navigation shouldn't depend on sim packages and sim packages shouldn't depend on navigation as well. They should be more atomic and ultimately provide a demo package with integrated launch files.

JesusSilvaUtrera, 2024-03-21:
I will take care of this issue

JesusSilvaUtrera, 2024-03-25:
@francocipollone @garyservin I have been looking into the issue, and I want to share my thoughts to see if you agree.

We should use the `bringup.launch.py` file for either gazebo simulation or real robot (we have the `use_sim_time` and other params to change according to the necesities) and move the `andino_simulation_navigation.launch.py` to the `andino_gz_classic` package (or even delete it, as this will cause the gazebo package to depend on the navigation package, which is also not ideal).

At the end, we should see that thse kind of packages should be independent of the others, as they are some kind of 'add-ons', we can launch the simulation without having the navigation part and we can navigate without having the simulation part.

This way we would delete the 'andino_gz_classic' and the 'turtlebot3_gazebo' dependencies from package.xml.

We would obviously also need to update the documentation of the package according to the new structure.

JesusSilvaUtrera, 2024-04-09:
@francocipollone I have been digging a bit into this and trying a few things, and there are some topics I would like to discuss:

- There is a known issue that if it is the first time that you are launching Gazebo in the container, it sometimes doesn't launch well. It has an easy solution, you have to source the 'Gazebo classic configuration' after sourcing the workspace: `source /usr/share/gazebo/setup.bash`. I think it would be good adding this as a note in the `andino_gz_classic` README.
- Instead of just deleting the `andino_simulation_navigation.launch.py`, we could add a new package (e.g: `andino_examples` or `andino_apps`) with complete use cases that make use of the other packages.
- For simulation, we should add namespace support, because there could be times where we have multiple robots at once (the `bringup.launch.py` already has this support, we would need to add it in `andino_gz_classic`).

Let me know what do you think when you have the time, thanks in advance!

PR #235 (2024-04-19): andino_apps package created and andino_navigation package updated
# 🦟 Bug fix

Fixes #228 

## Summary
I have created a new `andino_apps` package, where I moved the `andino_simulation_navigation` launch file from `andino_navigation`. This makes the `andino_navigation` independent of the simulation packages.

## Test it
Build the workspace and follow the new README.md from `andino_apps` to test the launch file and see that the functionality remains the same.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as neede [... cut]
closed 2024-04-19 by agalbachicar