Ekumen-OS/andino #236: Add support for the Bitcraze™ Lighthouse Positioning System

hidmic, 2024-04-25:
## Desired behavior

The [Lighthouse Positioning System](https://www.bitcraze.io/documentation/system/positioning/ligthouse-positioning-system/) by Bitcraze is a low-cost pose estimation system for their flagship product, the Crazyflie MAV. It uses a number of IR beacons and a receiver on a deck (the equivalent of Arduino shields) to provide the Crazyflie MAV with pose estimates in real-time.

If we could leverage this deck, we could provide Andino with pose tracking support. This is immensely useful when validating, tuning, and benchmarking localization, mapping, and navigation systems. Also when teaching about them.

## Alternatives considered

There are other, full blown motion capture solutions in the market but these are exceedingly expensive when compared with Andino's target budget. We are talking tens of thousands of US dollars.

## Implementation suggestion

The one difficulty with this is that both the hardware interface and the communication protocol are very loosely documented. The pinout for the [Lighthouse Positioning deck](https://www.bitcraze.io/documentation/hardware/lighthouse_deck/lighthouse_deck-datasheet.pdf) and for [Crazyflie expansion decks](https://www.bitcraze.io/documentation/system/platform/cf2-expansiondecks/) in general is public. The firmware that talks to this deck is also open source (see [here](https://github.com/bitcraze/crazyflie-firmware/tree/d9ee34a78cecf2eaf8b05e0b54487cff0c7b4590/src/modules/src/lighthouse) and [here](https://github.com/bitcraze/crazyflie-firmware/tree/d9ee34a78cecf2eaf8b05e0b54487cff0c7b4590/src/utils/src/lighthouse)). But the protocol itself is nowhere to be found. It can only be reversed engineered. It shouldn't be difficult to do so for one of the typical on-board computers for Andino (e.g. an RPi4).

hidmic, 2024-04-25:
> But the protocol itself is nowhere to be found. It can only be reversed engineered.

Aha! This may not be 100% true. We shall take a look at some of the documentation and tools over at https://github.com/bitcraze/lighthouse-fpga. Thanks @glpuga.

JCarosella, 2024-04-27:
Some info here, found the [Bitcrazy API for deck drivers](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/userguides/deck/). This will help understand the src code linked above for the lighthouse driver code

JCarosella, 2024-05-02:
## Status update:

I was able to hook up the board to a RPi4 and start receiving data from the sensors with which the positioning algorithm could be implemented.

The bad news: I mistakenly connected the VCC/GND pins backwards and most likely burnt something. Most likely because the board turns on and responds to commands, but does not output sensor info (I think [the TS4231 IR receivers](https://www.bitcraze.io/documentation/hardware/lighthouse_deck/lighthouse_deck-revd.pdf) burnt).

I will be getting a replacement, but that puts the project on hold for a while (no US market, only Europe). I'll keep this issue updated.