Ekumen-OS/andino #239: Update connections diagram

JesusSilvaUtrera, 2024-04-30:
## Desired behavior
The connections diagram doesn't include all the required parts and it is out-of-date, so it would be nice to update it with the latest changes to ease the assembly process.

## Implementation suggestion
Several considerations:
- The encoders and the motors are the same piece, so they should be together.
- The Raspberry is not included in the diagram
- Add the power part well explained, replacing the potato.
- The Arduino UNO should be replaced by an Arduino Nano (optional but the most used is the Nano for this robot)
- Add the LiDAR (optional as its wiring is not something too hard)

## Additional context

JesusSilvaUtrera, 2024-05-08:
To keep the issue updated, I am creating 3 diagrams in Google Drawings, one with the Arduino connections, the second with Raspberry Pi connections and the third for Power connections. I think this way is easier to follow the diagrams and then just connects the different diagrams into the final robot.

They won't be finished until I test them assembling my own robot.

francocipollone, 2024-05-08:
> To keep the issue updated, I am creating 3 diagrams in Google Drawings, one with the Arduino connections, the second with Raspberry Pi connections and the third for Power connections. I think this [...]

Sounds good!

PR #252 (2024-05-28): Update connections diagram
# 🎉 New feature

Closes #239 

## Summary
I have updated the Arduino diagram with the new voltages for the power, and I added the Raspberry diagram, which also includes the step-up and the IMU connections. It is in the `andino_hardware/README.md` file.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)

**Note to maintainers**: Remember to use **Squash-Merge** and edit the commit message to match the pull request summary while retaining `Signed-off-by` messages.
closed 2024-05-28 by jballoffet