Ekumen-OS/andino #271: Try out Raspberry Pi 5 with Andino

JesusSilvaUtrera, 2024-09-23:
## Summary

With the migration to Jazzy, a new problem appeared. `Ubuntu Mate 24` is apparently not going to be released, so we tried out with Ubuntu Server 24 on the Raspberry Pi 4, but the experience is not the same.

However, after doing some research it was found that `Ubuntu Mate 24` is not going to be released because the new Raspberry Pi 5 supports Ubuntu Desktop, performing quite well.

This issue is being opened to explore the possibility to buy a Raspberry Pi 5 and try it out with the Andino and `Ubuntu Desktop 24.04`, to see if the performance is good.

If the experiment is successful, we should use the Raspberry Pi 5 as default from now on, given that its price is similar to the Rapsberry Pi 4.

JesusSilvaUtrera, 2024-10-24:
**UPDATE**: Ubuntu Mate 24 was finally released, so we can try it out with the new Raspberry Pi 5.

[Release notes](https://ubuntu-mate.org/blog/ubuntu-mate-noble-numbat-release-notes/)

JesusSilvaUtrera, 2024-10-24:
It would be good to test Raspberry Pi 5 with Ubuntu Desktop 24 (it doesn't support Ubuntu Desktop 22) and Raspberry Pi 4 with Ubuntu Mate 24

Russ76, 2026-03-08:
I have installed Andino on the Raspberry Pi 5 with Ros Jazzy. Most things seem to work fine, except for the robot state publisher. It only publishes one line, not the several items it published about in Humble. There is no camera link, no second base, no laser, no wheels, etc. I can't find the problem causing this so far. I am not using Mate, just the server version of Ubuntu 24.04