Ekumen-OS/andino #272: FAIL TO 'colcon build' THE FIRST TIME

thanhtruongphan, 2024-10-27:
Hello everyone! 👋

I just tried to download andino_repositories, and follow the README.md instructions to install. 

But I have some problems when I use _**colcon build**_ 

1. When I ```colcon build``` the first time: 
>>> [image]

I solved by ```sudo apt install ros-humble-ament-cmake-clang-format```

2. When I ```colcon build``` the second time: 
>>>[image]

I solved by ```sudo apt install libserial-dev```

3. When I ```colcon build``` the third time: 
>>>[image]

I solved by ```sudo apt-get install libgflags-dev```

4. When I ```colcon build``` the fourth time: 
>>>[image]

... and now I don't know how to solve it. 

Help me, pls.

Thank you all so much, and have a nice day! 🥹

thanhtruongphan, 2024-10-27:
UPDATE: 

I tried to mode the line 12 and It's OK now.
>>[image]

result: 
>>>>>[image]

thanhtruongphan, 2024-10-27:
and in SBC: 
[image]

francocipollone, 2024-10-27:
Hey @thanhtruongphan , thanks for the issue creation. 

1. I feel like the issues with the stuff not installed is due to some missing rosdep install: e.g: typically `rosdep install --from-paths src -y -i`

2. Regarding the issues and the modifications to the CMakeLists file you are doing. It is odd and not happening on my end last time a built everything (I am going to try asap when I return from a trip). Could you share your setup? e.g: OS / hardware/ etc

thanhtruongphan, 2024-10-28:
yeah! 
you're right! 

after ```sudo rosdep init``` and ```rosdep update```, I forget to ```rosdep install --from-paths src -y -i``` 😄

probably because of the errors in the previous project that made me stupid.

anyway, thank you so much for your reply.

have a nice day 🥹

closed 2024-10-31 by thanhtruongphan