Ekumen-OS/andino #274: HOW TO RUN ON REAL ROBOT

thanhtruongphan, 2024-10-28:
I've set up all firmware both on SBC and dev-PC, 👋

I could run gazebo with teleop-twist-keyboard successfully, 

but the Real-robot doesn't work...
I don't know how to do...

Can anyone help me, pls...

Thank you so much and have a nice day!! 🥹

>>> ![Screenshot 2024-10-28 214557](https://github.cI've set up all firmware both on SBC and dev-PC, 👋

I could run gazebo with teleop-twist-keyboard successfully, 

but the Real-robot doesn't work...
I don't know how to do...

Can anyone help me, pls...

Thank you so much and have a nice day!! 🥹

>>> [image]
om/user-attachments/assets/7fb6fad4-d259-4203-9086-bdbc40ba6baf)

thanhtruongphan, 2024-10-29:
I tried with the data 'o 255 255' to serial but nothing happen...
>>> [image]

thanhtruongphan, 2024-11-02:
oh, oh, maybe Ubuntu on VMvirtualbox machine can connect properly with SBC, 

now I can use teleop_twist_keyboard to control two motors. 

but, maybe I have some problem, 
**when I run command ```ros2 launch andino_bringup andino_robot.launch.py```**:

[image]

more detail:
```
[INFO] [launch]: All log files can be found below /home/truongpt/.ros/log/2024-11-02-11-05-19-391756-truongpt-sbc-96943
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [97024]
[robot_state_publisher-1] [WARN] [1730520322.065477803] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-1] [INFO] [1730520322.071813321] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1730520322.074996867] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1730520322.075066514] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-1] [INFO] [1730520322.075105976] [robot_state_publisher]: got segment caster_base_link
[robot_state_publisher-1] [INFO] [1730520322.075141049] [robot_state_publisher]: got segment caster_rotation_link
[robot_state_publisher-1] [INFO] [1730520322.075173660] [robot_state_publisher]: got segment caster_wheel_link
[robot_state_publisher-1] [INFO] [1730520322.075205585] [robot_state_publisher]: got segment left_motor
[robot_state_publisher-1] [INFO] [1730520322.075237066] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-1] [INFO] [1730520322.075268898] [robot_state_publisher]: got segment right_motor
[robot_state_publisher-1] [INFO] [1730520322.075300379] [robot_state_publisher]: got segment right_wheel
[robot_state_publisher-1] [INFO] [1730520322.075331786] [robot_state_publisher]: got segment rplidar_laser_link
[robot_state_publisher-1] [INFO] [1730520322.075363415] [robot_state_publisher]: got segment second_base_link
[INFO] [v4l2_camera_node-2]: process started with pid [97086]
[INFO] [rplidar_composition-3]: process started with pid [97105]
[v4l2_camera_node-2] [INFO] [1730520325.915706520] [v4l2_camera]: Driver: bm2835 mmal
[v4l2_camera_node-2] [INFO] [1730520325.916024772] [v4l2_camera]: Version: 331683
[v4l2_camera_node-2] [INFO] [1730520325.916058753] [v4l2_camera]: Device: mmal service 16.1
[v4l2_camera_node-2] [INFO] [1730520325.916080919] [v4l2_camera]: Location: platform:bcm2835-v4l2-0
[v4l2_camera_node-2] [INFO] [1730520325.916101271] [v4l2_camera]: Capabilities:
[v4l2_camera_node-2] [INFO] [1730520325.916121659] [v4l2_camera]:   Read/write: YES
[v4l2_camera_node-2] [INFO] [1730520325.916141955] [v4l2_camera]:   Streaming: YES
[v4l2_camera_node-2] [INFO] [1730520325.916184195] [v4l2_camera]: Current pixel format: YUYV @ 640x480
[v4l2_camera_node-2] [INFO] [1730520325.917353745] [v4l2_camera]: Available pixel formats: 
[v4l2_camera_node-2] [INFO] [1730520325.917410170] [v4l2_camera]:   YU12 - Planar YUV 4:2:0
[v4l2_camera_node-2] [INFO] [1730520325.917433188] [v4l2_camera]:   YUYV - YUYV 4:2:2
[v4l2_camera_node-2] [INFO] [1730520325.917452965] [v4l2_camera]:   RGB3 - 24-bit RGB 8-8-8
[v4l2_camera_node-2] [INFO] [1730520325.917472632] [v4l2_camera]:   JPEG - JFIF JPEG
[v4l2_camera_node-2] [INFO] [1730520325.917491909] [v4l2_camera]:   H264 - H.264
[v4l2_camera_node-2] [INFO] [1730520325.917511131] [v4l2_camera]:   MJPG - Motion-JPEG
[v4l2_camera_node-2] [INFO] [1730520325.917530279] [v4l2_camera]:   YVYU - YVYU 4:2:2
[v4l2_camera_node-2] [INFO] [1730520325.917549315] [v4l2_camera]:   VYUY - VYUY 4:2:2
[v4l2_camera_node-2] [INFO] [1730520325.917568648] [v4l2_camera]:   UYVY - UYVY 4:2:2
[v4l2_camera_node-2] [INFO] [1730520325.917587481] [v4l2_camera]:   NV12 - Y/CbCr 4:2:0
[v4l2_camera_node-2] [INFO] [1730520325.917606629] [v4l2_camera]:   BGR3 - 24-bit BGR 8-8-8
[v4l2_camera_node-2] [INFO] [1730520325.917625591] [v4l2_camera]:   YV12 - Planar YVU 4:2:0
[v4l2_camera_node-2] [INFO] [1730520325.917644887] [v4l2_camera]:   NV21 - Y/CrCb 4:2:0
[v4l2_camera_node-2] [INFO] [1730520325.917664294] [v4l2_camera]:   RX24 - 32-bit XBGR 8-8-8-8
[v4l2_camera_node-2] [INFO] [1730520325.917684109] [v4l2_camera]: Available controls: 
[v4l2_camera_node-2] [INFO] [1730520325.917723200] [v4l2_camera]:   Brightness (1) = 50
[v4l2_camera_node-2] [INFO] [1730520325.917758366] [v4l2_camera]:   Contrast (1) = 0
[v4l2_camera_node-2] [INFO] [1730520325.917790606] [v4l2_camera]:   Saturation (1) = 0
[v4l2_camera_node-2] [INFO] [1730520325.917822847] [v4l2_camera]:   Red Balance (1) = 1000
[v4l2_camera_node-2] [INFO] [1730520325.917853290] [v4l2_camera]:   Blue Balance (1) = 1000
[v4l2_camera_node-2] [INFO] [1730520325.917882790] [v4l2_camera]:   Horizontal Flip (2) = 0
[v4l2_camera_node-2] [INFO] [1730520325.917912104] [v4l2_camera]:   Vertical Flip (2) = 0
[v4l2_camera_node-2] [INFO] [1730520325.917942326] [v4l2_camera]:   Power Line Frequency (3) = 1
[v4l2_camera_node-2] [INFO] [1730520325.917972473] [v4l2_camera]:   Sharpness (1) = 0
[v4l2_camera_node-2] [INFO] [1730520325.918002287] [v4l2_camera]:   Color Effects (3) = 0
[v4l2_camera_node-2] [INFO] [1730520325.918031935] [v4l2_camera]:   Rotate (1) = 0
[v4l2_camera_node-2] [INFO] [1730520325.918061416] [v4l2_camera]:   Color Effects, CbCr (1) = 32896
[... cut ...]
[spawner-6] [INFO] [1730520338.641385975] [spawner_diff_controller]: Loaded diff_controller
[ros2_control_node-4] [INFO] [1730520338.662279010] [controller_manager]: Configuring controller 'diff_controller'
[spawner-6] [INFO] [1730520338.789621437] [spawner_diff_controller]: Configured and activated diff_controller
[INFO] [spawner-6]: process has finished cleanly [pid 97314]
```

**and when I run command to open slam package ```ros2 launch andino_slam slam_toolbox_online_async.launch.py```**

[image]

and no Rviz or any GUI is show up...

thanhtruongphan, 2024-11-02:
and when I run command line to open rviz after biringup **```ros2 launch andino_bringup rviz.launch.py```**

[image]

francocipollone, 2024-11-04:
Hey there @thanhtruongphan , thanks for your interest.

Let's try to understand the context, identify the issue, and then narrow down the problem that you might be facing. 

0. If you want to use the simulation at some point, check [andino_gz](https://github.com/Ekumen-OS/andino_gz). That's the recommended simulation (gz_classic is outdated). 

1. Were you able to move the real robot? Assuming for the output of the andino_bringup launch it seems that everything is alright, at least no errors were shown.

2. andino_slam's launch file by default is not launching any rivz or GUI visualization ( imagine that you can either launch that on an external laptop or directly in the SBC). You have a `ros2 launch andino_slam rviz.launch.py` for launching only rviz with some convenient setup when doing slam.

3. That issue you mentioned when running "ros2 launch andino_bringup rviz.launch.py" might be related most probably to the fact that `map` is set as fixed frame and the map is not being published. Try changing the fixed frame to base_link and see if you actually see the robot and the wheels are rotating when you rotate the real wheels (to fully see that your set up is ok and for example encoders are correctly working)

thanhtruongphan, 2024-11-04:
hi @francocipollone very happy and thank for your reply. and have a nice day. :1st_place_medal: 

now, I have problem with work between SBC-computer and ubuntu laptop. Both is Ubuntu 22.04 and ROS2 humble like the instructions in README.md file, of course Ubuntu mate on Raspi4. 

The issue is I can't use the "discovery" feature https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html#try-some-examples

I tried https://docs.ros.org/en/humble/Tutorials/Beginner-CLI-Tools/Configuring-ROS2-Environment.html#the-ros-domain-id-variable but it doesn't work.

I can't listen on SBC from Ubuntu laptop or listen on Ubuntu laptop from SBC. So, I when I run Rviz on Ubuntu laptop, it doesn't show up anything. (I installed Ubuntu dual boot with Windows11 and of course use both Ubuntu on one Wifi)

Have you ever saw that problem? I've reinstalled Ubuntu mate on my SBC and hope it work... :smiling_face_with_tear:

francocipollone, 2024-11-04:
> The issue is I can't use the "discovery" feature https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html#try-some-examples

Run a "talker" node in your laptop and and `listener` node on your SBC and vice-versa. See if the communication goes both ways.

Verify that you are not using any ROS_LOCALHOST or ROS_DOMAIN_ID env var on both computers. Later on, if you want to isolate the robot in a particular ROS_DOMAIN_ID you can do it. However if it is your first time doing so in ROS 2 start easy.

If everything is ok and still not working you can see if your router is actually causing some issue. So try using another network: The easiest way is to use your smartphone to set up a wifi network real quick and connect laptop and sbc to that wifi network and see what happens.

thanhtruongphan, 2024-11-04:
yeah, here is env var on mine 

>>> [image]

I'm reinstalling ubuntu mate on SBC, and I will test again... :jack_o_lantern:

thanhtruongphan, 2024-11-04:
yeah, using Wifi from smartphone helped me to solve problem "talker/listener". Great~ @francocipollone 

now when I run **```rosdep install --from-paths src -i -y -r```**

[image]

result: 
[image]

I don't care it then continue the instructions to install, and finally run the command line **```ros2 launch andino_bringup andino_robot.launch.py```**

it auto close and the error is: 

[image]

help me, pls :1st_place_medal: 
-thank you-

thanhtruongphan, 2024-11-04:
camera and lidar and arduino is on, successfully check in https://github.com/Ekumen-OS/andino/tree/humble/andino_hardware :smiling_face_with_tear:

thanhtruongphan, 2024-11-04:
yeah... the problem has been solved by using Wifi share by smartphone. 

Thank you so much~ :100:

thanhtruongphan, 2024-11-04:
sorry, i'm new to ubuntu and ROS2.

can i ask a question?

can i use this project on Jetson nano devkit (B01)?
I have a Jetson nano devkit and a research project that works with this Jetson and ROS (ROS2 in case of andino) and uses machine vision with some image processing tasks.

-thanks-

francocipollone, 2024-11-04:
That issue that was shown in the resolved comment seems something with the camera, were you able to solve it? Do you have a camera installed?

> can i use this project on Jetson nano devkit (B01)?

You definitely can, I haven't tried this there but it should be fully compatible code-wise, you might need to make some changes at some point I imagine. However if you go that path it would be great to come back and mention how your experience with Jetson nano was when using it in a Andino.