Ekumen-OS/andino #276: Arduino code compilation issue

maxxlife, 2024-11-08:
When I try to compile Arduino ino file project from andino/andino_firmware, receive an error. Do you have a list of libraries to add to the Arduino IDE environment?
```
In file included from /home/maxxlife/andino/andino_firmware/src/app.h:32:0,
                 from /home/maxxlife/andino/andino_firmware/src/app.cpp:65:
/home/maxxlife/Arduino/libraries/Adafruit_BNO055/Adafruit_BNO055.h:29:10: fatal error: Adafruit_Sensor.h: No such file or directory
 #include <Adafruit_Sensor.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

francocipollone, 2024-11-08:
Check the readme at `andino_firmware's` package: https://github.com/Ekumen-OS/andino/tree/humble/andino_firmware

maxxlife, 2024-11-09:
> Check the readme at `andino_firmware's` package: https://github.com/Ekumen-OS/andino/tree/humble/andino_firmware
> 
> There you will see the path for loading it using arduino IDE or platform io.

Thanks, using terminal command pio I was able to compile it. Could I build a project for Arduino Leonardo?

francocipollone, 2024-12-20:
> > Check the readme at `andino_firmware's` package: https://github.com/Ekumen-OS/andino/tree/humble/andino_firmware
> > There you will see the path for loading it using arduino IDE or platform io.
> 
> Thanks, using terminal command pio I was able to compile it. Could I build a project for Arduino Leonardo?

Hey @maxxlife to be honest I haven't tried it with Leonardo. The pinout should match and regarding the interruptions leonardo has a couple of hardware interruptions that are switched, however we mainly use software interruptions in the code. So it SHOULD work from theory.
In case you build it for Leonardo please feel free to comment here with the outcome! All info is welcomed! :D

Closing the issue as it is resolved.

closed 2024-12-20 by francocipollone

maxxlife, 2024-12-20:
Yes, I was able to build for Leonardo. There is difference in Interrupts names. But it can be reconfigured following the datasheets for Leonardo and Arduino Uno different ATMega controllers. I compiled the code for Leonardo with the updated Interrupt name, but didn't test it.

francocipollone, 2024-12-20:
> Yes, I was able to build for Leonardo. There is difference in Interrupts names. But it can be reconfigured following the datasheets for Leonardo and Arduino Uno different ATMega controllers. I compiled the code for Leonardo with the updated Interrupt name, but didn't test it.

Thanks for closing the loop on that. I appreciate the information :D

maxxlife, 2024-12-20:
> Thanks for closing the loop on that. I appreciate the information :D

I will create a PR or feature for that. Currently I'm in the heavy job search process, and also lost paper with all my notes about Interrupts reconfigure during my home move.